SignalK/instrumentpanel #3: React-based rewrite

tkurki, 2015-02-23:
Rewrite UI and grid logic with React.

tkurki, 2015-02-23:
@bobmor99 I have a 'new, improved' version of instrumentpanel available in a branch.

It is not really ready for public consumption, but if you want you can give it a go by installing it with `bower install https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui`

If you choose to try it I would appreciate feedback on what works and what doesn't.

bobmor99, 2015-02-23:
@tkurki,

Comments at the bottom.

Installation showed this:
~/signalk-server-node$ bower install
https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui
bower not-cached
https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui#*
bower resolve
https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui#*
bower download
https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui
bower extract       react-grid-ui#* react-grid-ui
bower resolved
https://github.com/SignalK/instrumentpanel/zipball/react-grid-ui#e-tag:2aeabb9c0
bower not-cached    git://github.com/twbs/bootstrap.git#~3.3.1
bower resolve       git://github.com/twbs/bootstrap.git#~3.3.1
bower download      https://github.com/twbs/bootstrap/archive/v3.3.2.tar.gz
bower extract       bootstrap#~3.3.1 archive.tar.gz
bower resolved      git://github.com/twbs/bootstrap.git#3.3.2

Unable to find a suitable version for bootstrap, please choose one:
    1) bootstrap#~3.2.0 which resolved to 3.2.0 and is required by
simplegauges#af4625673b
    2) bootstrap#~3.3.1 which resolved to 3.3.2 and is required by
instrumentpanel#e-tag:2aeabb9c0Prefix the choice with ! to persist it to
bower.json

? Answer:: !2
bower resolution    Saved bootstrap#~3.3.1 as resolution
bower install       instrumentpanel#e-tag:2aeabb9c0
bower install       bootstrap#3.3.2
bower no-json       No bower.json file to save to, use bower init to create
one

instrumentpanel#e-tag:2aeabb9c0 bower_components/instrumentpanel
├── bootstrap#3.3.2
├── font-awesome#4.2.0
├── jquery#2.1.3
└── leaflet#0.7.3

bootstrap#3.3.2 bower_components/bootstrap
└── jquery#2.1.3

---------------------------------------------------------------------------------------------

~/signalk-server-node$ ls -l bow*
total 52
drwxrwxr-x 7 bobmor99 bobmor99 4096 Feb 23 14:32 bootstrap
drwxrwxr-x 6 bobmor99 bobmor99 4096 Feb 23 08:43 components-font-awesome
drwxrwxr-x 2 bobmor99 bobmor99 4096 Feb 23 08:19 d3
drwxrwxr-x 6 bobmor99 bobmor99 4096 Feb 23 08:19 font-awesome
drwxrwxr-x 5 bobmor99 bobmor99 4096 Feb 23 08:19 gridster
drwxrwxr-x 4 bobmor99 bobmor99 4096 Feb 23 14:32 instrumentpanel
drwxrwxr-x 4 bobmor99 bobmor99 4096 Feb 23 08:19 jquery
drwxrwxr-x 3 bobmor99 bobmor99 4096 Feb 23 08:19 leaflet
drwxrwxr-x 2 bobmor99 bobmor99 4096 Feb 23 08:43 L.EasyButton
drwxrwxr-x 3 bobmor99 bobmor99 4096 Feb 23 08:43 maptracker
drwxrwxr-x 4 bobmor99 bobmor99 4096 Feb 23 08:20 sailgauge
drwxrwxr-x 4 bobmor99 bobmor99 4096 Feb 23 12:35 simplegauges
drwxrwxr-x 5 bobmor99 bobmor99 4096 Feb 23 08:44 socket.io-client

-------------------------------------------------------------------------------------------

1) I like the Settings checkboxes for selecting gauges to include/exclude.
2) It would be nice to reintroduce the sizing capability. e.g.
environment.date, environment.time and navigation.position do not fit in
their boxes.
3) navigation.speedThroughWater now has an appropriate amount of precision.
4) Gauge names are very hard to read (too small). Maybe use some sort of
label instead of the SK element name.

All for now.
Good stuff!

tkurki, 2015-02-23:
Thanks!

Resize is still there, it's just not working with the bower installed version, need to fix that.

Bootstrap version conflict I fixed in simplegauges.

The idea for labels is have nicer human readable ones, so we'll need a mapping like https://github.com/SignalK/instrumentpanel/blob/structured-widgets/lib/signalknames.js (just a few as an example).

tkurki, 2015-04-24:
Done.

closed 2015-04-24 by tkurki