SignalK/instrumentpanel #4: Java server compatibility

tkurki, 2015-04-09:
Instrumentpanel (both current master & react version in branch) are incompatible with the current java server.

rob42, 2015-04-09:
Problem is related to lack of session and associated transport type being set in the java server. Hence the java server output is never sent back. This is now fixed in the java server.

Instrumentpanel does not send a subscribe message on connection, so the java server only sends a heartbeat message anyway.

rob42, 2015-04-09:
Subscribe added in pull request #5
Now shows everything, including sources!

tkurki, 2015-04-10:
With 4df39402eeff620d4a923aee0955b8d0abf91176 ignores paths not in schema, so @rob42's extra source stuff stays hidden.

closed 2015-04-10 by tkurki