SignalK/instrumentpanel #8: re-size gauge fail under chrome

manuprendlair, 2015-04-27:
Hello,

I had problem to resize gauge.
Exactly when I lock the layout the new size of the gauge is loss and go back to previous.

My config:
Chrome Version 42.0.2311.90 m under windows 7.

after connecting to the server, I had 4 gauges.
I select 2: SOG and wind.
I unlock layout.
resize the wind (bigger)
resize the SOG. (bigger)
then lock the layout -> the SOG go back to the initial size.

I could repeat the problem.
I I try with firefox v29.0.1 it work without any pb...

Best Regards

tkurki, 2016-02-16:
Is this problem still valid?

manuprendlair, 2016-02-17:
Hello Teppo.

Long time I didnt test it because I am in my third years of nurse learning
and havent much time to work on this.
The project is not dead just in standby.
Did you make modifications on code since I write about this problem?
Sorry actualy I am not able to test it. I need to restart my raspberry
config to test it and no time for that now.
You could close it and when I will work again on it I will make another
issue.

Best Regards

closed 2016-02-17 by tkurki