SignalK/instrumentpanel #13: Version of InstrumentPanel in Demo section of new website is not the latest

sumps, 2015-08-19:
The new website is great and the Demo section allows anyone to "touch and feel" Signal K. However, what we present in the Demo section has to be the latest and greatest, so please could the Demo consumers on the Signal K site be kept up to date ? 

InstrumentPanel in particular is not up to date and when you open it does not look its best....

[image]

timmathews, 2015-08-19:
I've updated demo.signalk.org to the latest version, however I think if you've never visited before it will still default to the view you posted as the layout is saved in your browser via LocalStorage.

sumps, 2015-08-19:
Hi Tim, yes I realise this and I had intentionally opened a new incognito window so that I could see what a new visitor to the site would see.

Thanks for updating the demo and let's hope we get lots of views.

closed 2016-02-07 by tkurki