SignalK/instrumentpanel #17: Handle multiple sources

tkurki, 2016-01-09:
IP currently has no notion of multiple sources for a single path. It should be able to present the difference.

sumps, 2016-03-17:
This does need a solution and I suggest opening extra gauges for each source. Yesterday I had NMEA2000 logged depth and NMEA0183 simulated depth coming in and the single gauge for Depth Below Transducer in Instrument Panel was alternating between the two values. This is not good behaviour and it would be better to have two depth gauges one for each source and if there was a method to turn off some gauges then it would be an even better solution.

tkurki, 2016-03-17:
This strategy works for gauges that show a single value. 

For complex gauges that present multiple values in a single gauge we need something more complex, like a config ui to pick the source.

tkurki, 2016-05-01:
Initial support is there. Windmeter should be updated with a configuration interface allowing you to specify the source related logic.

closed 2019-10-23 by tkurki