SignalK/instrumentpanel #19: Cannot resize or arrange gauges in Browser

sumps, 2016-02-08:
I am using the latest stand alone version of Instrument Panel and it appears to be working perfectly except I cannot resize or arrange the gauges.
Going in to Settings I can turn off/on the display of certain panels, but going in to Padlock, the panels turn yellow, the resize button is  visible in the lower right corner of the adjustable panels and the cursor changes to a diagonal arrow when placed over it but I cannot click and drag to move or resize in Chrome or Win10 Edge. However, in Safari on my iPad I can resize and arrange OK.

tkurki, 2016-02-08:
I can not reproduce this. 

Anything appearing in the JavaScript console?

rob42, 2016-02-09:
I was using the earlier version on Firefox this weekend and had the same problem, running in a tablet with android 4.4 (I think). Worked ok in chrome.

closed 2016-02-16 by tkurki