SignalK/instrumentpanel #30: Better visual design

tkurki, 2016-05-15:
The current InstrumentPanel is pretty utilitarian - a bit of visual design would not hurt, including a night palette.

lauriboren, 2016-05-29:
Here's a quick take on a visual design. It's not perfect, but it's something. And if someone wants to implement this, I'm more than happy to help.

A couple of notes:

- The design is based on a twelve column grid. Elements could then be 2/3/4/6 columns width.
- The type is [Open Sans by Google](https://www.google.com/fonts/specimen/Open+Sans).
- I included the 40x40px icons in 2x and 3x.

The boat blueprint (bottom right module) could include true/apparent wind direction and at least main sail boom position.

[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

timmathews, 2016-05-29:
That looks really good! A couple of suggestions. I'd like to be able to name the pages (e.g. "Engine", "Electrical", etc), and I think the menu ought be hidden most of the time.

We also need to be able to handle alarms for shallow water, approaching AIS contacts, bilge pump running, etc. Some sort of dismissible  non-modal alert (like jQuery Growl). Or maybe we should be using desktop notifications?

What do you need from us to make this happen? I'm happy to help!

timmathews, 2016-05-29:
@lauriboren, did you join our Slack? http://slack.signalk.org/