SignalK/instrumentpanel #31: Handle ws and http on different ports

rob42, 2016-06-20:
Connect to  localhost:8080
22:07:51.595 Firefox can't establish a connection to the server at ws://localhost:8080/signalk/v1/stream?subscribe=self 

Connect to  localhost:3000
GET http://localhost:3000/signalk/v1/stream  (should be ws://)

It should initially connect to the http port, eg localhost:8080/signalk and get the endpoints page. Then use that to find the ws port and connect.
Not sure where the GET http:// is called from, cant see it in the code, but I think its part of getting the self value.

tkurki, 2017-02-24:
Do we have a permanently running demo version of java server available? I believe this is fixed, but don't have java server handy right now to verify.

http://node-master.signalk.org/instrumentpanel/ has a fairly recent version of IP (freshly installed whenever SK node server master gets autodeployed there).

rob42, 2017-02-24:
Its sometimes up at http://www.42.co.nz:8088, its there now, but demo data doesnt seem to be streaming...
Will check into it later

tkurki, 2019-10-23:
Closing as inactive.

closed 2019-10-23 by tkurki