SignalK/instrumentpanel #37: Feedback from Users to make IP easier to use

sumps, 2017-02-16:
Feedback from iKommunicate users that have used Instrument Panel is that once setup it is very useul BUT that setting it up is time consuming and the initial "out of the box" experience could be much improved with some simple changes;

1)  First time that IP is run (or after a reset) that the user is taken straight in to the Settings "By Display Widget" page and all gauges have Visible unticked, so that the user chooses the gauges to turn on.

2)  The default units for all angular measurements are set to Degrees

3)  When you change a distance unit to Meters, a temperature unit to Celsius, etc. you are given the option to change all units of the same type to that new value i.e. change temperature once and all temperatures are changed to the same unit type.

4)  Rather than one long single column of gauges, by default they are grouped in to columns i.e. Column 1 all of the Environment keys, Column 2 all of the Navigation keys, Column 3 all of the Propulsion.0 keys, Column 4 all of the Propulsion.1 keys, etc. This will possibly remove the need for the user to shuffle the gauges around or at least make it quicker and easier.

5)  All time related keys to be displayed in a more user friendly format i.e. Hours:Minutes:Seconds rather than seconds and all log type distances to be shown in NM, KM or SM rather than meters.

6)  Wind Gauge to have a Windspeed Units selection.

I think that is everything.

tkurki, 2017-02-24:
Let's see how people feel about this with #45 / #40 in place.

sumps, 2017-02-28:
I agree - this is looking less useful now, with some of the other issues sorted.

cmotelet, 2020-02-19:
Closed by #126

closed 2020-02-19 by cmotelet