SignalK/instrumentpanel #41: User Feedback on IP #5 of 6 - "More user friendly time and distance displays"

sumps, 2017-02-16:
Feedback from iKommunicate users that have used Instrument Panel is that once setup it is very useul BUT that setting it up is time consuming and the initial "out of the box" experience could be much improved with six simple changes;

5)  All time related keys to be displayed in a more user friendly format i.e. Hours:Minutes:Seconds rather than seconds and all log type distances to be shown in NM, KM or SM rather than meters.

tkurki, 2017-02-24:
These two, or at least the distances one requires that IP knows which keys are distances - unit being meter is not enough. Adding that information to the schema and making it available on the http interface as part of the meta is not really a huge task. IP could include the data clientside as a fallback.

Hardcoding some keys to be for example distance in IP is not really a good idea.

See https://github.com/SignalK/specification/issues/130.

cmotelet, 2019-10-23:
For time value, I think it's solved by `feature: add digitaldatetime component` in #66
For distances it is more complicated and I agree with tkurky.
For example, for me:
- drift in m/s
- design.​length in m
- environment.​depth.​belowKeel in m
- but navigation.​log in nm
In a future version, I will think about the possibility of proposing several optional templates with different mapping path => unit in the spirit of `preferred units`.

cmotelet, 2020-01-02:
closed by #66 & #103

closed 2020-01-02 by cmotelet