SignalK/instrumentpanel #47: Some Temperatures have no Unit Selection

sumps, 2017-02-26:
All of the propulsion temperatures default to Kelvin and have no option to change them.

tkurki, 2017-03-01:
What exact paths are you having problems with? The ones that I tested work ok:

[image]

sumps, 2017-03-01:
OK this is the latest Instrument Panel (defaults after reset) with data from iKommunicate. I note at least one non-compliant key but perhaps you can see more...

[image]

sumps, 2017-03-01:
I see that we are using propulsion.x.engineTemperature and the spec is just propulsion.x.temperature and we also have propulsion.x.oilTemp when it should be oilTemperature.

OK looks like an iKommunicate bug, will get this sorted.

sumps, 2017-03-01:
However, the Batteries temperature issue is a different "kettle of fish". It looks like a whole series of keys have been added for battery temperature alarm limits. These surely are metadata and not measured values ? I think this has got messed up at some fairly recent point, as I am sure it has not always been like this.

The measured value is the battery temperature and this should be electrical.batteries.x.temperature which is what iKommunicate is outputting.

I think this needs fixing in the schema.

tkurki, 2017-03-01:
The telltale is the missing unit (in your picture): then IP can not find the unit from the schema and will just show the value as it is.

tkurki, 2017-03-01:
Yes, I noticed also that the limits that should be in metadata are messed up as data keys.

They should be like https://github.com/SignalK/specification/blob/master/schemas/groups/electrical.json#L65-L74

but are 
https://github.com/SignalK/specification/blob/master/schemas/groups/electrical.json#L147-L151

tkurki, 2019-10-23:
Closing as out of date.

closed 2019-10-23 by tkurki