SignalK/instrumentpanel #55: Fonts are too big in firefox

rob42, 2017-04-03:
Open the instrumentpanel in firefox. The fonts are too big and overlap. I looked into the css with dev tools, but could not see the cause  or affect the display to any extent.

tkurki, 2017-04-03:
Could you be a bit more specific? What data, what widget, what widget size?

This is what you get from http://node-master.signalk.org/instrumentpanel/  out of the box (I just moved position so that it is shown and not below the fold):

[image]

Position is not pretty, but there is no generic _fonts are too big and overlap_ problem afai can see.

As the final result is a combination of individual panel's aspect ratio and `text`s' font sizes it is not a question of adjusting font sizes for just one case.

For example changing the position widget's aspect ratio by reducing height one notch results in 
[image]

rob42, 2017-04-03:
Its on all the panels, I didnt specify as it seems to be a firebox thing. Looks normal in chrome

rob42, 2017-04-03:
Firefox thing. <curse>cell phone </curse>

timmathews, 2017-06-02:
FWIW, this is how the latest version renders in Firefox 53 on Windows.

[image]

No amount of resizing seems to help

closed 2019-10-23 by tkurki