SignalK/instrumentpanel #56: UI should allow optional units for propulsion.port.fuel.rate

aplathan, 2017-05-04:
Currently only m3/s is available for propulsion.port.fuel.rate. Because the real life unit is so small, it gets rounded up to 0.00, and is useless. Users should be able to choose commonly used units such as L/h the same way temperatures allow K, C, F and so on.

tkurki, 2017-05-06:
This should be fixe in master now. Let me know if you run into problems.

closed 2017-05-06 by tkurki

aplathan, 2017-05-06:
Not sure if the fix is in master. I can't get the updated files with npm install signalk/instrumentpanel. Apologies in advance if I'm doing yet another rookie mistake with npm.

tkurki, 2017-05-06:
My mistake, committed to wrong branch. Will fix later...

tkurki, 2017-05-07:
Now it hopefully works. The latest version is deployed at http://node-master.signalk.org/instrumentpanel/ , you can try it out from there and connect to your server that has relevant data items.

commit 3646926a (2017-05-06): feature: add conversions for m3/s

Fixes #56.
commit f657bbae (2017-05-07): feature: add conversions for m3/s

Fixes #56.