SignalK/instrumentpanel #57: Additional Pressure Units

ph1l, 2017-05-09:
Request Submitted from mailing list user:

> for many of the numbers different units can be set. For pressure there is 
> Pa, psi, Bar and kPa, but not the two most common, namely mbar and hPa.
> 
> Old timers might want to see mmHg or torr, but this is just a curiosity. 
> How do I add hPa and mbar as extra formats and associated formulas ?
>

tkurki, 2017-05-10:
New conversions are fairly easy to add, you just have to make sure that the underlying conversion library, js-quantities, supports the unit in question.

See https://github.com/SignalK/instrumentpanel/commit/f657bbae66dcd59656aac4cc1b5f117eefacefed

olewsaa, 2017-05-10:
The definitions.js contains all the units including mmHg and torr, and all prefixes including Hecto/h so hPa should be no problem. I just do not in which source file I need to update.

tkurki, 2017-05-10:
Fixed. The relevant file was linked in my previous comment, but you need to be able to build the React app.

You can upgrade to the latest version with `npm install SignalK/instrumentpanel` in the Node server directory or try it out at http://localhost:3000/instrumentpanel/ and connect to your server that produces some data with Pascals as unit.