SignalK/instrumentpanel #58: Build issues

cschmitt2000, 2017-06-07:
I would like to try to make some changes to the UI.  But when I  run "npm run build"  I get the following errors on both Raspberry PI and on my Windows machines.  Running Node 7.x  pulling from master

ERROR in ./~/signalk-client/test/url.js
Module not found: Error: Cannot resolve module 'chai' in  .....\instrumentpanel\node_modules\signalk-client\test
 @ ./~/signalk-client/test/url.js 1:11-26

ERROR in ui.js from UglifyJs
SyntaxError: Unexpected token: operator (>) [./~/signalk-schema/src/fullsignalk.
js:112,0]

tkurki, 2017-06-11:
The culprit for the first error is the dynamic `require`s in signalk-client related to optionally having `mdns` present, which is never there in browser environment anyway.

I am not sure about the best way to resolve this and don't have time to investigate right now. If you want to get ahead you can just replace mdns require with undefined at 
https://github.com/SignalK/signalk-js-client/blob/master/index.js#L156 and return false in the test at https://github.com/SignalK/signalk-js-client/blob/master/index.js#L362.

With no dynamic requires webpack does not need to create a context for runtime resolution of requires, where lack of chai becomes an issue. 

Alternatively you can install the dev dependencies for signalk-client, which should make the error go away.

I created https://github.com/SignalK/signalk-js-client/issues/23 to track the real issue.

CBSailor, 2017-08-15:
Teppo,
I am running into the same, the module reported the issue on is main.js in my case.
Unfortunately none of the suggested fixes made any change. All did result in the same output. 
It starts failing with the HTML code in main.js in row 45.

Looks like someone was able to compile it at some point. So I may be missing something here. 
I'd appreciate any assistance. 
Thanks
Chris
`> webpack -p

Hash: cb2052f4664649e79904
Version: webpack 1.15.0
Time: 136ms
   [0] multi ui 28 bytes {0} [built] [1 error]
    + 1 hidden modules

ERROR in ./lib/ui/main.js
Module parse failed: C:\signalk-server-node\node_modules\instrumentpanel\lib\ui\
main.js Unexpected token (45:6)
You may need an appropriate loader to handle this file type.
SyntaxError: Unexpected token (45:6)
    at Parser.pp$4.raise (C:\signalk-servernode\node_modules\instrumentpanel\node_modules\webpack\node_modules\acorn\dist\acorn.js:2221:15)
    at Parser.pp.unexpected (C:\signalk-server-node\node_modules\instrumentpanel\node_modules\webpack\node_modules\acorn\dist\acorn.js:603:10)
....
`

tkurki, 2019-10-23:
Probably outdated, closing.

closed 2019-10-23 by tkurki