SignalK/instrumentpanel #59: Red section on analog also at too low values

olewsaa, 2017-08-14:
Today one can set a red section at the high end of the scale. I would like to see that also for the low end, like fresh water where it's should be red at some low level. Another option would be green when it's ok and red when it's not ok, like the example of voltage red from 0-10 V, green from 11-15, and red from 16 and higher (correspondingly higher for those of you who have 24V system.

Ole
https://sites.google.com/site/olewsaa/yacht-server-with-raspberry

tkurki, 2017-08-14:
I started implementation, so that you could configure zones for values in your SK server, and display them in InstrumentPanel as you describe, but I got sidetracked and it stalled: 

https://github.com/SignalK/instrumentpanel/tree/zones-support

rob42, 2017-08-14:
In signalk you can define a zone (in meta) for a low value as you would a high value. so you can have 
```
 "meta": {
                    "displayName": "Depth Below Keel",
                    "longName": "Depth Below Keel in Meters",
                    "shortName": "DBK",
                    "gaugeType": "sparkLine",
                    "units": "Meters (m)",
                    "warnMethod": ["visual"],
                    "alarmMethod": ["sound"],
                    "zones": [
                        {"lower":0.0,"upper":1.5,"state":"alarm", "message":"Running aground!"},
                        {"lower":1.5,"upper":3.0,"state":"warn", "message":"Shallow water!"},
                        {"lower":3.0, "upper":100.0, "state":"normal"},
                        {"lower":100.0, "state":"alarm", "message": "Scary, its really deep here"}
                    ]
                }
```

rob42, 2017-08-14:
BTW nice work on your nav project :-)

Stuga, 2020-09-21:
I would also like this feature for fuel level.  Looking at the code, it should be pretty simple to implement in _analogcomponent.js_, by replicating the _redArc()_ function.

Google have something similar in https://developers.google.com/chart/interactive/docs/gallery/gauge, where you can define 3 zones "Green", "Yellow", "Red" with their own ranges.  You can redefine a discrete colour for each if you want, so you can redefine Yellow to be '#ff0000' (red)  and have two red bands on the same gauge.

I have no experience with Node, else I could help!

Thanks

cmotelet, 2020-09-26:
@Stuga ,
It is true that the addition of several color zones has been requested for a long time. We talked about linking it directly to the zone in the server so that we don't have to configure them on the client. This option was a bit dormant. Linked to the zones or with a local configuration, I have to add the code in the widget to display several colors.
I'll put this project back on top of the stack ;-)

Stuga, 2020-09-30:
That's great Christian.

I look forward to this functionality :-)

Stuart

cmotelet, 2020-11-22:
the feature is present in version 17.0.0
[image]

closed 2020-11-22 by cmotelet