SignalK/instrumentpanel #62: Instrument panel does not update boolean values true / false

Krillle, 2018-03-06:
It seems to me the Instrumenpanel 0.2.6 is not updating keys with the boolean values true and false. They seem to be initialized with the value delivered by the second related PGN message and stay at that value. 

Test data:
```
2014-08-15T19:00:00.152,2,65280,0,255,8,30,99,01,00,00,00,00,00
2014-08-15T19:00:05.085,2,65280,0,255,8,30,99,01,f4,01,e8,03,ff
2014-08-15T19:00:10.659,2,65280,0,255,8,30,99,01,00,00,0,00,00
```

Test Plugin:
https://github.com/sbender9/signalk-empirbus-plugin

Test keys:
```
"electrical": {
    "switches": {
      "empirBusNxt-instance0-dimmer1": {
        "state": {
          "value": true,
          "$source": "signalk-empirbus-nxt",
          "timestamp": "2018-03-04T21:52:33.013Z"
        },
        "dimmingLevel": {
          "value": 0.5,
          "$source": "signalk-empirbus-nxt",
          "timestamp": "2018-03-04T21:52:33.013Z"
        }
      }
```
While dimmingLevel is updated every 5 seconds, the state is not and stays true.
First PNG at 2014-08-15T19:00:00.152 seems to be ignored.

tkurki, 2018-03-08:
Any errors in console?

Krillle, 2018-03-08:
Yes, this one:
```
TypeError: undefined is not an object (evaluating 'this.state.value.toString')
exports_render — ui.js:33279
_renderValidatedComponentWithoutOwnerOrContext — ui.js:50786
_renderValidatedComponent — ui.js:50793
_updateRenderedComponent — ui.js:50770
_performComponentUpdate — ui.js:50765:242
updateComponent — ui.js:50743:352
performUpdateIfNecessary — ui.js:50734:230
performUpdateIfNecessary — ui.js:1539
s — ui.js:1051
perform — ui.js:7990:95
perform — ui.js:7990:95
T — ui.js:1105
a
closeAll — ui.js:8025:87
perform — ui.js:7998
batchedUpdates — ui.js:34225
u — ui.js:1057:86
r — ui.js:12657
enqueueSetState — ui.js:12699
setState — ui.js:33989:469
http://signalk:3000/
[native code]
http://signalk:3000/
http://signalk:3000/
n — ui.js:2029
h — ui.js:2089
push — ui.js:2516
handleEvent — ui.js:2537
http://signalk:3000/
push — ui.js:3115
push — ui.js:46623:99
http://signalk:3000/
forEach
http://signalk:3000/
forEach
handleDelta — ui.js:46616
dispatch — ui.js:12226
[native code]
onmessage — ui.js:5540
```

Krillle, 2018-03-08:
I specified issue description:
It seems to me the Instrumenpanel 0.2.6 is not updating boolean keys to false. They seem to be initialized with the first true value stay at that value.

tkurki, 2018-03-08:
Nevertheless fixed in v0.2.7

closed 2018-03-08 by tkurki