SignalK/instrumentpanel #72: Strange Pitch roll decoding?

dmdelorme, 2019-01-16:
`I am getting a text box for Pitch/ Roll/Yaw..
With the following... 

> {  "pitch": -0.0401425739, "roll": -0.114929934405,"yaw": 0}

This is the json

"attitude": {
      "meta": {
        "description": "Vessel attitude: roll, pitch and yaw"
      },
      "value": {
        "pitch": -0.0401425739,
        "roll": -0.114929934405,
        "yaw": 0
      },
      "$source": "OPsensors.I2C.MPU-925x",
      "timestamp": "2019-01-16T22:03:40.984Z"
    }

`Should it not break the component values out;
navigation/attitude/pitch
navigation/attitude/roll
Ditch yaw as we have headingmagnetic with the proper value

cmotelet, 2019-01-17:
There is no specific widget for this SK path in Instrument panel. It use the default generic plugin who display value content whithout decoding.
This is a very specific content not really readable by human.
I can add one dedicated widget in my next PR but displaying 3 float numbers is not really sexy.
Comments are welcome if someone have an idea for a good  display format.

rob42, 2019-01-17:
Could try a HUD diplay like an aircraft https://proxy.duckduckgo.com/iu/?u=http%3A%2F%2Faerotask.de%2Fwp-content%2Fuploads%2F2015%2F04%2FK%C3%BCnstlicher-Horizont.png&f=1

dmdelorme, 2019-01-18:
I think the crazy decimal places is a result of ujson. in openplotter and pypilot.
Is this an error in openplotter/pypilot I thought that I had Pitch and Roll before in the instrument panel.

cmotelet, 2019-01-29:
#76  minimal widget who display yaw/pitch/roll is now integrated in new version of InstrumentPanel.
[image]

closed 2019-02-03 by dmdelorme