SignalK/instrumentpanel #98: change discritoin text ?

dl6dbh, 2019-05-02:
hallo 
is it possible to change the discription text in the panels ? 
e.g. 
enviroment.water.temperture   into    wassertemp

cmotelet, 2019-05-02:
Sorry but right now, yes and no.
instrumentpanel support `displayName` meta but it is not yet functional for the paths already existing in the signalk schemas.
I think it will be functional in a future version, we need to make changes in the signalk server to be able to do it.
I am working on this feature:
https://github.com/SignalK/signalk-server-node/issues/754#issue-423429006
https://github.com/SignalK/signalk-server-node/pull/773#issue-270101088

dl6dbh, 2019-05-03:
Hallo Christian,
Thanks for answer. Will wait to Future.

cmotelet, 2020-01-19:
With version 1.20.0 of signalk-server-node you can now add the `displayName` key to the server's `default.json` file.
```
      "environment": {
        "water": {
          "temperature": {
            "meta": {
              "displayName": "wassertemp"
            }
          }
        }
```
Restart your server and clear the `InstrumentPanel` cache by adding `flushCache=true` at the end of your URL (see instrument panel help online: 5. Trouble).
In the next version of `instrumentPanel` I will add an option to flush the cache through the GUI.

closed 2020-01-19 by cmotelet

MatthewBastian, 2020-04-11:
Not so sure its working as advertised. I add eight displayName changes to defaults.json and it only accepted the last one I entered? Can only one be renamed or is this a case of finger trouble?

{
  "vessels": {
    "self": {
      "uuid": "urn:mrn:signalk:uuid:e60307c1-d5f3-4b88-8075-323b954b87d5",
      "name": "Juma",
      "design": {
        "draft": {
          "value": {
            "maximum": 1
          }
        },
        "length": {
          "value": {
            "overall": 11.3
          }
        },
        "beam": {
          "value": 3.7
        },
        "airHeight": {
          "value": 15.25
        },
        "aisShipType": {
          "value": {
            "name": "Sailing",
            "id": 36
          }
        }
      },
      "sensors": {
        "gps": {
          "fromBow": {
            "value": 7
          },
          "fromCenter": {
            "value": 2
          }
        }
      },
      "electrical": {
        "batteries": {
          "house": {
            "voltage": {
              "meta": {
                "displayName": "House Battery"
              }
            }
          }
        }
      },
      "electrical": {
        "batteries": {
          "starter": {
            "voltage": {
              "meta": {
                "displayName": "Sarter Battery"
              }
            }
          }
        }
      },
      "electrical": {
        "alternators": {
          "3QM30": {
            "temperature": {
              "meta": {
                "displayName": "Alternator"
              }
            }
          }
        }
      },
      "propulsion": {
        "3QM30": {
          "CoolantTemperature": {
            "meta": {
              "displayName": "Coolant"
            }
          }
        }
      },
      "propulsion": {
        "3QM30": {
          "transmission": {
            "OilTemperature": {
              "meta": {
                "displayName": "Transmission"
              }
            }
          }
        }
      },
      "propulsion": {
        "3QM30": {
          "temperature": {
            "meta": {
              "displayName": "Engine Block"
            }
          }
        }
      },
      "propulsion": {
        "3QM30": {
          "oilPressure": {
            "meta": {
              "displayName": "Oil"
            }
          }
        }
      }
    }
  }
}

cmotelet, 2020-04-11:
@MatthewBastian ,
You must first test if the server returns the right `meta` as follows (change path with your path):
GET http://yourSkServer:3000/signalk/v1/api/vessels/self/navigation/courseOverGroundTrue/meta
And check if return correct `meta` with your `displayName`:
```
{
  "units": "rad",
  "description": "Course over ground (true)",
  "displayName": "COGt"
}
```
If OK, you must flush `meta` stored in cache of your device :
http://yourSkServer:3000/@signalk/instrumentpanel/?flushCache=true
Since version 0.14.0, this option is integrated in the reset menu.
If you enable the console on your browser, you should see once the requests to the meta urls.

tkurki, 2020-04-11:
Your json has repeating elements, for example electrical should be there once. On my phone, so can’t provide a fixed version. The structure should be hierarchical.

MatthewBastian, 2020-04-11:
Christian and Teppo,

Thanks for the quick and insightful replies, these helped me solve my issues and they were all my finger trouble! In case someone else makes it here, I've added my /home/pi/.signalk/defaults.json file for someones future reference.

Back ground is that I just put together an engine monitoring system for my sailboat using an ESP8266 and MCP3008 chip combo and I'm looking for a way to display these on my Raspberry Pi. NMEA0183 is too dated (limited) in its ability to deal with this data and hence putting Signal K as the lower level data aggregater.

Thanks,

Matt

{
  "vessels": {
    "self": {
      "uuid": "urn:mrn:signalk:uuid:e60307c1-d5f3-4b88-8075-323b954b87d5",
      "name": "Juma",
      "design": {
        "draft": {
          "value": {
            "maximum": 1
          }
        },
        "length": {
          "value": {
            "overall": 11.3
          }
        },
        "beam": {
          "value": 3.7
        },
        "airHeight": {
          "value": 15.25
        },
        "aisShipType": {
          "value": {
            "name": "Sailing",
            "id": 36
          }
        }
      },
      "sensors": {
        "gps": {
          "fromBow": {
            "value": 7
          },
          "fromCenter": {
            "value": 2
          }
        }
      },
      "electrical": {
        "batteries": {
          "starter": {
            "voltage": {
              "meta": {
                "displayName": "Sarter Battery voltage"
              }
            }
          },
          "house": {
            "voltage": {
              "meta": {
                "displayName": "House Battery voltage"
              }
            }
          }
        },
        "alternators": {
          "3QM30": {
            "temperature": {
              "meta": {
                "displayName": "Alternator temp"
              }
            }
          }
        }
      },
      "environment": {
        "inside": {
          "engineRoom": {
            "temperature": {
              "meta": {
                "displayName": "Engine Room temp"
              }
            }
          }
        }
      },
      "propulsion": {
        "3QM30": {
          "coolantTemperature": {
            "meta": {
              "displayName": "Coolant temp"
            }
          },
          "temperature": {
            "meta": {
              "displayName": "Engine Block temp"
            }
          },
          "oilPressure": {
            "meta": {
              "displayName": "Oil press"
            }
          },
          "transmission": {
            "oilTemperature": {
              "meta": {
                "displayName": "Transmission temp"
              }
            }
          }
        }
      }
    }
  }
}