SignalK/instrumentpanel #109: Dark mode

bergie, 2019-09-16:
Many modern devices can support a dark mode with the [prefers-color-scheme](https://developer.mozilla.org/en-US/docs/Web/CSS/@media/prefers-color-scheme) media query. Adding dark styling to the instrument panel utilizing this would allow automatic switching to a more night vision friendly coloring at night time.

Quick-and-dirty option would be:

```css
@media (prefers-color-scheme: dark) {
  body {
    filter: invert(100%);
    background-color: #000;
  }
}
```

This would look like:
[image]

Of course one could design a nicer color scheme, too.

cmotelet, 2019-09-16:
I am glad to know that you use instrumentpanel also at night.
I had already thought about this option without knowing if it would be of interest to users.
I'm putting this feature at the top of the list. It will be added in the next version.

bergie, 2019-10-17:
BTW, the way Freeboard-SK did it is pretty nice: https://github.com/SignalK/freeboard-sk/blob/6ee94482d5982300aa34c4c1975a8734da4d24d0/CHANGELOG.md

- **Added:** Alternate color scheme (`Dark mode`) configuration setting with options to use either the: 
    1) Device OS setting via the `prefers-color-scheme` css media attribute
    2) Signal K `environment.mode` value.

cmotelet, 2019-10-17:
Thanks Bergie, I had seen and tested it as soon as it went online.
The dark mode for IP is in progress...
I work on css and cleaning hard coded colors.
I will most likely use the same options as `freeboard-sk` with maybe an extra button for the manual switching.
https://github.com/cmotelet/instrumentpanel/tree/darkMode-dev
[image]
[image]

cmotelet, 2020-01-02:
Closed by #118

closed 2020-01-02 by cmotelet