SignalK/instrumentpanel #124: Setting to determine whether new widgets should show automatically

mirthestam, 2020-01-01:
I would like to see my instrument panels continue to look the way I designed them. It is not pleasant when new widgets appear which I do not want to see on the panel at all.  One cannot anticipate all information that will later become available on the server. Removing this while sailing is awkward. A setting not to add them automatically would help with this.

mirthestam, 2020-01-01:
I have noticed that this is on the task list. Let me know if I can help.

cmotelet, 2020-01-01:
Ok, in the next release, I'll add a checkBox in the settings to make the new widget visible or not.
This checkBox will be checked by default.
And I have to do some cleaning up in the task list, it's been a long time since I updated it ;-)

tkurki, 2020-01-02:
The behavior that all new paths get automatically added is useful when debugging, so that you see everything you've got, but it is pretty confusing and even irritating as mirthestam commented.

If we make _don't add new items in the grid_ the default the user starts with *nothing visible*, which is also a bad experience, _why is this not showing anything_. 

Possible fixes:

1. always add the first item to the grid - maybe better than nothing, but not very useful, because what shows will be random
1. if there are no visible gauges add automatically position/speed*/depth/wind, whichever comes first
1. if there are no visible gauges go automatically to settings
1. if there are no visible gauges show a note saying _No gauges visible, please press /spannericon/ to configure your panel._

Comments?

mirthestam, 2020-01-02:
I see three use cases here.The first user experience, the configuration (i.e. debugging) and finally the use at sea.

It seems to me that the best user experience for new users is the most important. And with that the display of a simple panel. For example, consisting of the most common instruments such as the compass, windmeter and the speedometer. In addition, the above note to help the user on their way to configuring the panel.

A user who is familiar with the panel knows how to find his way to the settings and is quickly able to adjust the panel to his wishes after a clean install.

cmotelet, 2020-01-02:
I had well thought about the case of the first use, and that's why I specified "the checkBox will be checked by default" => new widget visible.

Then, the user will be able to choose not to display the new widgets anymore by unchecking the checkBox. By this we avoid the case of the empty page.

When the user unchecks the checkBox, a warning message will indicate that the new widgets will be set to not visible by default.

The display of alarms in the dedicated tab will not be subject to this checkBox.

Personally, I'm not for arbitrarily choosing which widgets should be added to the first page on the first use. Rather, I'm all about adding everything at the beginning and letting the user see the range of possible widgets.  And in my opinion adding code to filter widgets in the unique case of first use is useless. 

Let the user make his choices by simply pointing out the risks and benefits.

tkurki, 2020-01-02:
If you have a fairly well instrumented vessel the panel gets real crowded very quickly for example
- 3 solar panel controllers with V/A/W for input and output plus total/daily generated power, charging mode =>  18 gauges
- 2 GPS's  with about 9 paths => 18 gauges, most of which are "bullshit", like navigation.gnss.satellites and navigation.gnss.type

So for example on a phone the first experience is this total mess. This is why I have been thinking about just showing some basics by default.

mirthestam, 2020-01-02:
> So for example on a phone the first experience is this total mess. This is why I have been thinking about just showing some basics by default.

This is exactly what I have experienced. 

>  And in my opinion adding code to filter widgets in the unique case of first use is useless.

Agree, a filter is not the intention. the proposal is to add a few paths by default, preferably with the 'add widgets automatically' setting disabled.

I would not underestimate the importance of the first user experience. It may not be a common situation, but it is a crucial moment when people decide to use SignalK or not.

cmotelet, 2020-01-02:
So OK, I can start with a bypass on "position/speed/depth/wind" and all the other widgets hidden by default.
And in settings a "ToggleButtonGroup" add new discovered widgets on grid:
- Only "position/speed/depth/wind" (selected by default)
- All
- None

cmotelet, 2020-02-14:
closed by #126 , new release of package comming soon.
I still have a few features and fixes to add before.

closed 2020-02-14 by cmotelet