SignalK/instrumentpanel #127: Persist panel layouts on server

bergie, 2020-02-03:
Currently all instrumentpanel config is client-side, meaning that you need to do all the layouting, unit conversion prefs etc again every time you use a different device.

I think it'd be cool to persist the config and panel layouts server-side in a plugin config JSON file.

cmotelet, 2020-02-03:
You're right, a centralized backup on the server side is missing a lot. Several ways are being explored:
- Add a backup/restore menu to download the configuration in a single json file on client side.
- Switch instrumentpanel to server plugin mode to offer an individual API to backup / restore client configuration on server side.
- Used a common API on the SK server that is in Work in Progress.
I will probably start with my first proposal to switch after on second or third proposal.

bergie, 2020-02-03:
Great! For our e-ink instrument panel I went with a config file:
https://github.com/meri-imperiumi/dashboard/blob/master/config.py

Note: this also has the feature of switching between layouts automatically when vessel's navigation.state changes. So, for example showing one set of data when sailing, and another one when anchored

mirthestam, 2020-02-08:
I like to have different layouts per device. This combined with a way to store layouts at the server would be great.

cmotelet, 2020-05-25:
we are about to release the new version of Instrumentpanel with  backup on the server.
Can you try it and give me your feedback ?
The online help is updated with details on how to use the backup.
```
sudo su - pi (if pi is user running signalk)
cd .signalk
npm install git://github.com/SignalK/instrumentpanel#applicationData-beta
```

seaplusplus, 2020-07-18:
I tested this today and the load / save function worked perfectly. 

It it possible to implement an auto save / load function for a custom layouts?  I would love to be able to make changes on one device and have it just appear on another. 

Thanks for this, it is very much appreciated.

cmotelet, 2020-07-19:
@seaplusplus ,
We are thinking of a notification sent by the server when a new layout is saved.
At the moment, for autosave, I think it's a bad idea because when you switch from portrait to landscape, the layout changes.
And an autoreload on another device may have a negative effect if the client has changed the layout himself.
The layouts stored on the server are used as a template.
When you import one, it's stored in localstorage in the device's browser.
After the import, there is no more direct relationship between the layout stored on the server and the one used by the device.
And the layout lives its life alone on the device.

cmotelet, 2020-07-19:
closed via #152 & release 0.15.0

closed 2020-07-19 by cmotelet