SignalK/instrumentpanel #134: Being able to delete a useless widget

cmotelet, 2020-02-18:
Some widgets shown or hidden on the grid are no longer useful when you have removed the equipment that generated the Signal K path. The path related to the equipment is dead but the widget is still created when launching the panel because it is present in the layout.
There should be a way to delete them without resetting the layout.

EdKok, 2020-03-05:
Second that.

jeremyrainman, 2020-05-28:
Would this not be solved with a server-side grid configuration, or perhaps a persistent client-side grid configuration? I agree with the desire wholly.

Ianknx, 2020-12-06:
Hello

Are any alarm settings available I cannot seem to find any and have tried setting up zones but they do not seem to integrate.

Regards

Ian

cmotelet, 2020-12-06:
There is no settings available in InstrumentPanel execpt capability to hide an alarm.
InstrumentPanel only displays the contents of these paths : /signalk/v1/api/vessels/self/notifications.
A plugin like `signalk-simple-notifications` can help you to generate notification.

cmotelet, 2021-08-11:
closed by #193

closed 2021-08-11 by cmotelet