SignalK/instrumentpanel #173: Instrumentpanel doesn't load: 404 error getting "/signalk/v1/applicationData/global/instrumentpanel/1.0/instrumentpanelPreferredUnits"

ivan-ns, 2020-12-31:
After last update Instrumentpanel doesn't load. 
Server response 404 for '/signalk/v1/applicationData/global/instrumentpanel/1.0/instrumentpanelPreferredUnits'

cmotelet, 2020-12-31:
can you give me a little more information on 'Instrumentpanel doesn't load' ?
- A totally white screen ?
- A white screen with only '0/X shown path on grid  & Signal K Instrument Panel version 0.17.1' ?
- Other ?

which version of
- InstrumentPanel?
- Signal K server ?
- browser ?
- OS ?

`Server response 404 for '/signalk/v1/applicationData/global/instrumentpanel/1.0/instrumentpanelPreferredUnits'` is not an error, it indicates that there is no data stored for the preferred units on the server.

ivan-ns, 2020-12-31:
@cmotelet , thanks for your answer.
- Yes, white screen with only "Signal K Instrument Panel version 0.17.1" at the bottom.
The following html content
`<html><head>
    <meta charset="utf-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="initial-scale=1">
    <meta name="apple-mobile-web-app-capable" content="yes">
    <meta name="apple-mobile-web-app-status-bar-style" content="default">
    <meta name="mobile-web-app-capable" content="yes">
    <meta name="apple-mobile-web-app-title" content="SK-Instruments">
    <meta name="application-name" content="SK-Instruments">
    <meta name="msapplication-TileColor" content="#da532c">
    <meta name="theme-color" content="#ffffff">
    <title>SignalK Instrumentpanel</title>
    <link href="./css/bootstrap.min.css" rel="stylesheet">
    <link href="./css/instrumentpanel.css" rel="stylesheet">
    <link rel="apple-touch-icon" sizes="180x180" href="icons/apple-touch-icon.png">
    <link rel="icon" type="image/png" sizes="32x32" href="icons/favicon-32x32.png">
    <link rel="icon" type="image/png" sizes="192x192" href="icons/android-chrome-192x192.png">
    <link rel="icon" type="image/png" sizes="16x16" href="icons/favicon-16x16.png">
    <link rel="manifest" href="site.webmanifest">
    <link rel="mask-icon" href="icons/safari-pinned-tab.svg" color="#5bbad5">

  </head>

  <body>
    <div id="content" onclick=""></div>
    <footer style="text-align:center">
      <hr>
      <p><a href="https://github.com/SignalK/instrumentpanel">Signal K Instrument Panel</a> version <span id="version">0.17.1</span></p>
    </footer>
  

  <script src="ui.js"></script>

</body></html>`

Versions:
- Instrument Panel version 0.17.1
- Signal K server version 1.37.4
- Browser chrome 87.0.4280.88
- OS Windows 7 SP1
At the browser console I have
`/signalk/v1/applicationData/global/instrumentpanel/1.0/instrumentpanelPreferredUnits:1 Failed to load resource: the server responded with a status of 404 (Not Found)
react-dom.production.min.js:196 TypeError: Cannot read property 'name' of null
    at t.renderFunction (specifictext.js:302)
    at t.value (specifictext.js:71)
    at Ya (react-dom.production.min.js:181)
    at $a (react-dom.production.min.js:180)
    at ks (react-dom.production.min.js:261)
    at mu (react-dom.production.min.js:230)
    at hu (react-dom.production.min.js:229)
    at iu (react-dom.production.min.js:223)
    at react-dom.production.min.js:121
    at t.unstable_runWithPriority (scheduler.production.min.js:18)
fs @ react-dom.production.min.js:196
react-dom.production.min.js:121 Uncaught TypeError: Cannot read property 'name' of null
    at t.renderFunction (specifictext.js:302)
    at t.value (specifictext.js:71)
    at Ya (react-dom.production.min.js:181)
    at $a (react-dom.production.min.js:180)
    at ks (react-dom.production.min.js:261)
    at mu (react-dom.production.min.js:230)
    at hu (react-dom.production.min.js:229)
    at iu (react-dom.production.min.js:223)
    at react-dom.production.min.js:121
    at t.unstable_runWithPriority (scheduler.production.min.js:18)
/signalk/v1/api/vessels/self/resources/routes/urn:mrn:signalk:uuid:2a5b414f-515d-4cb9-bde8-30de44058906/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.routes.urn:mrn:signalk:uuid:2a5b414f-515d-4cb9-bde8-30de44058906
(anonymous) @ signalkschemadata.js:51
/signalk/v1/api/vessels/self/resources/waypoints/urn:mrn:signalk:uuid:e4efb9bc-c4bb-48be-aa01-9aaac5cf845e/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.waypoints.urn:mrn:signalk:uuid:e4efb9bc-c4bb-48be-aa01-9aaac5cf845e
(anonymous) @ signalkschemadata.js:51
/signalk/v1/api/vessels/self/resources/routes/urn:mrn:signalk:uuid:4983e6a0-54c6-489f-a5d9-13985489030f/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.routes.urn:mrn:signalk:uuid:4983e6a0-54c6-489f-a5d9-13985489030f
(anonymous) @ signalkschemadata.js:51
/signalk/v1/api/vessels/self/resources/routes/urn:mrn:signalk:uuid:e32a72ca-7efd-4477-af7d-eade5332dc47/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.routes.urn:mrn:signalk:uuid:e32a72ca-7efd-4477-af7d-eade5332dc47
(anonymous) @ signalkschemadata.js:51
/signalk/v1/api/vessels/self/resources/routes/urn:mrn:signalk:uuid:ebe3a1d1-2517-4501-bc03-b638935ee5de/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.routes.urn:mrn:signalk:uuid:ebe3a1d1-2517-4501-bc03-b638935ee5de
(anonymous) @ signalkschemadata.js:51
/signalk/v1/api/vessels/self/resources/routes/urn:mrn:signalk:uuid:587d3f42-7c94-4608-8cbb-c7d8d764f96b/meta/:1 Failed to load resource: the server responded with a status of 404 (Not Found)
signalkschemadata.js:51 No such meta for resources.routes.urn:mrn:signalk:uuid:587d3f42-7c94-4608-8cbb-c7d8d764f96b
(anonymous) @ signalkschemadata.js:51

`

cmotelet, 2020-12-31:
Sorry but I can't reproduce your problem.
Can you give me the content of this URL to see all the data produced by the server : /signalk/v1/api/vessels/self/

The `404 (Not Found)` are normal when the server does not provide meta for certain paths.

In the page do you still have the toolbar at launch and heart blinking ?
[image]
If yes, in settings mode, do you have any widgets displayed ?
If yes, click on the `Show all paths` button

ivan-ns, 2020-12-31:
No toolbar available at the page.
That's the content of /signalk/v1/api/vessels/self/:
{"uuid":"urn:mrn:signalk:uuid:70c826aa-5c2f-4f84-854a-b12c01abe70a","name":"Takushi","communication":null,"design":{"draft":{"meta":{"description":"The draft of the vessel","properties":{"minimum":{"description":"The minimum draft of the vessel","type":"number","units":"m"},"maximum":{"description":"The maximum draft of the vessel","type":"number","units":"m"},"current":{"description":"The current draft of the vessel","type":"number","units":"m"},"canoe":{"description":"The draft of the vessel without protrusions such as keel, centerboard, rudder","type":"number","units":"m"}}},"value":{"maximum":1.5},"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"},"length":{"meta":{"description":"The various lengths of the vessel","properties":{"overall":{"type":"number","description":"Length overall","units":"m"},"hull":{"type":"number","description":"Length of hull","units":"m"},"waterline":{"type":"number","description":"Length at waterline","units":"m"}}},"value":{"overall":12},"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"},"beam":{"meta":{"units":"m","description":"Beam length"},"value":3.5,"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"},"airHeight":{"meta":{"units":"m","description":"Total height of the vessel"},"value":15,"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"},"aisShipType":{"meta":{"description":"The ais ship type see http://www.bosunsmate.org/ais/message5.php","properties":{"id":{"type":"number","description":"The ship type number"},"name":{"type":"string","description":"The ship type name"}}},"value":null,"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"}},"sensors":{"gps":{"fromBow":{"meta":{"description":"The distance from the bow to the sensor location"},"value":6,"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"},"fromCenter":{"meta":{"description":"The distance from the centerline to the sensor location, -ve to starboard, +ve to port"},"value":0,"$source":"defaults","timestamp":"2020-12-31T14:54:10.345Z"}}},"navigation":{"courseGreatCircle":{"activeRoute":{"href":{"meta":{"description":"A reference (URL) to the presently active route, in resources."},"value":"/resources/routes/urn:mrn:signalk:uuid:587d3f42-7c94-4608-8cbb-c7d8d764f96b","$source":"freeboard-sk-helper","timestamp":"2020-12-31T15:03:10.804Z"},"startTime":{"meta":{"description":"The time this route was activated"},"value":"2020-12-22T13:51:26.503Z","$source":"freeboard-sk-helper","timestamp":"2020-12-31T15:03:10.804Z"}},"nextPoint":{"position":{"meta":{},"value":{"latitude":60.03707416822979,"longitude":29.398040771484375},"$source":"freeboard-sk-helper","timestamp":"2020-12-31T15:03:10.804Z"},"arrivalCircle":{"meta":{},"value":926,"$source":"freeboard-sk-helper","timestamp":"2020-12-31T15:03:10.804Z"},"distance":{"meta":{},"value":7006.60523606077,"$source":"derived-data","timestamp":"2020-12-31T15:03:31.594Z"},"bearingTrue":{"meta":{},"value":-0.9409873657216355,"$source":"derived-data","timestamp":"2020-12-31T15:03:31.594Z"}},"previousPoint":{"position":{"meta":{"description":"The position of lastPoint in two dimensions","properties":{"longitude":{"type":"number","description":"Longitude","units":"deg","example":4.98765245},"latitude":{"type":"number","description":"Latitude","units":"deg","example":52.0987654},"altitude":{"type":"number","description":"Altitude","units":"m"}}},"value":{"latitude":60.082726170912245,"longitude":29.974464636770996},"$source":"freeboard-sk-helper","timestamp":"2020-12-31T15:03:10.804Z"},"distance":{"meta":{"units":"m","description":"The distance in meters between previousPoint and the vessel's present position"},"value":27905.666464321283,"$source":"derived-data","timestamp":"2020-12-31T15:03:31.594Z"}},"crossTrackError":{"meta":{"units":"m","description":"The distance from the vessel's present position to the closest point on a line (track) between previousPoint and nextPoint. A negative number indicates that the vessel is currently to the left of this line (and thus must steer right to compensate), a positive number means the vessel is to the right of the line (steer left to compensate)."},"value":-4977.153637185911,"$source":"derived-data","timestamp":"2020-12-31T15:03:31.594Z"},"bearingTrackTrue":{"meta":{"units":"rad","description":"The bearing of a line between previousPoint and nextPoint, relative to true north."},"value":-1.7238086552881826,"$source":"derived-data","timestamp":"2020-12-31T15:03:31.594Z"}},"position":{"meta":{"description":"The position of the vessel in 2 or 3 dimensions (WGS84 datum)","properties":{"longitude":{"type":"number","description":"Longitude","units":"deg","example":4.98765245},"latitude":{"type":"number","description":"Latitude","units":"deg","example":52.0987654},"altitude":{"type":"number","description":"Altitude","units":"m"}}},"value":{"longitude":29.5,"latitude":60},"$source":"Vessel Sim.127","timestamp":"2020-12-31T15:03:31.592Z","pgn":129029},"datetime":{"meta":{"description":"Time and Date from the GNSS Positioning System"},"value":"2019-02-17T16:29:28Z","$source":"Vessel Sim.127","timestamp":"2020-12-31T15:03:31.592Z","pgn":129029},"gnss":{"antennaAltitude":{"meta":{"units":"m","description":"Altitude of antenna"},"value":148.94,"$source":"Vessel Sim.127","timestamp":"2020-12-31T15:03:31.592Z","pgn":129029},"satellites":{"meta":{"description":"Number of satellites"},"value":0,"$source":"Vesse [... cut ...]  Normal"},"$source":"Vessel Sim.17","timestamp":"2020-12-31T15:03:31.609Z","pgn":127489},"neutralStartProtect":{"meta":{},"value":{"state":"normal","method":["visual"],"message":"Port Neutral Start Protect is Normal"},"$source":"Vessel Sim.17","timestamp":"2020-12-31T15:03:31.609Z","pgn":127489},"shuttingDown":{"meta":{},"value":{"state":"normal","method":["visual"],"message":"Port Engine Shutting Down is Normal"},"$source":"Vessel Sim.17","timestamp":"2020-12-31T15:03:31.609Z","pgn":127489}}}}}

ivan-ns, 2020-12-31:
Sorry for disturbing at the NY, I think I found the solution:
I have changed in Server>Settings field "Ship Type" from "Wing in Ground" (I had it by default) to "Sailing" and instrument panel came alive! Happy New Year! SignalK is awesome!

cmotelet, 2020-12-31:
Great and very interesting bug:
In your data related to "Ship Type":
```
   "aisShipType": {
      "meta": {
        "description": "The ais ship type see http://www.bosunsmate.org/ais/message5.php",
        "properties": {
          "id": {
            "type": "number",
            "description": "The ship type number"
          },
          "name": {
            "type": "string",
            "description": "The ship type name"
          }
        }
      },
      "value": null,
      "$source": "defaults",
      "timestamp": "2020-12-31T14:54:10.345Z"
    }
```
Your value is null !!!

In my data it's an object:
```
"aisShipType": {
"meta": {
"description": "The ais ship type see http://www.bosunsmate.org/ais/message5.php",
"properties": {
"id": {
"type": "number",
"description": "The ship type number"
},
"name": {
"type": "string",
"description": "The ship type name"
}
}
},
"value": {
"name": "Wing In Ground",
"id": 20
},
"$source": "defaults",
"timestamp": "2020-12-31T15:21:28.327Z"
}
```
or 
```
"aisShipType": {
"meta": {
"description": "The ais ship type see http://www.bosunsmate.org/ais/message5.php",
"properties": {
"id": {
"type": "number",
"description": "The ship type number"
},
"name": {
"type": "string",
"description": "The ship type name"
}
}
},
"value": {
"name": "Sailing",
"id": 36
},
"$source": "defaults",
"timestamp": "2020-12-31T15:26:17.732Z"
}
```
I have to try to reproduce this case to see if that's what it is then protect InstrumentPannel against it.
And happy new year ;-)

tkurki, 2020-12-31:
New meta deltas have a problem that creates nulls. Just noticed it yesterday, in aisreporter bug reports, @sbender9 said he’d look into it.

ivan-ns, 2020-12-31:
Sorry, it's offtopic, but I'm new in SignalK. I'm using 'activeRoute' in freeboard-sk, is there any way to automatically switch to next waypoint when one is reached? (notifications.arrivalCircleEntered fired)

tkurki, 2021-01-02:
Please use https://github.com/SignalK/freeboard-sk/ for Freeboard stuff.

closed 2021-02-09 by tkurki