SignalK/instrumentpanel #174: Not really an "issue"

wmbutler, 2020-12-31:
Sorry for leaving this comment here, but I spent some time this weekend reviewing all the signalk projects and I'm pretty impressed by the community. I've been in the dev world for a few decades and am interested in contributing to this repo, specifically in the area of custom skins for instrument panels.

As I reviewed your code, it is apparent that you've spent a lot of time and energy on your widgets. I think there is a great opportunity to break out the svg contents from the js code to allow for a library of compatible skins to be developed. For example:

https://github.com/SignalK/instrumentpanel/blob/b229e0beeab8a920e57cd59c142f3e5b36fe42d4/lib/ui/widgets/digitalcomponent.js#L49

Here you have an svg embedded in the widget `digitalcomponent.js`. I think it would be really cool for this to be a variable that references `/lib/ui/skins/default/digitalcomponent.svg`. Contributors could then design their own skins inside `/lib/ui/skins/`.

Some other things I noticed in yours svg code is that there appear to be some discrepancies in syntax that I'd normally expect. For instance in this svg your syntax is `textAnchor` instead of `text-anchor`. I'm also a little rusty on the difference between `text-anchor` and `dominant-baseline`. 

```
      <svg height="100%" width="100%" viewBox="0 0 20 33" stroke="none">
        <text x="10" y="4" textAnchor="middle" fontSize="6" dominantBaseline="middle">
          {this.props.options.label}
        </text>
        <text textAnchor="end" x="60" y="30" fontSize={unitFontSize} dominantBaseline="baseline">{unit}</text>
        <text textAnchor="middle" x="10" y="22" fontSize="29" dominantBaseline="middle">
          {this.state.value.toString()}
        </text>
      </svg>
```

I was also curious about your choice of `viewBox="0 0 20 33"`. Was this arbitrary? I think basing these numbers on 12 column for eventual flex support might be interesting.

Last, I see that your widgets display the dotted notation for the various streams being received. Do you envision that instrumentpanel should be responsible for crosswalking those values to more readable values (i.e. `environment.depth.belowTransducer` - `Depth`) or should that crosswalk be performed at the signalK server?

Thanks for answering all these questions and I'm looking forward to contributing.

tkurki, 2021-01-02:
Thanks for your comments!

It appears that you are not familiar with React and jsx. The `camelCase` naming pattern is the way hyphenated attributes are handled in jsx. Furthermore the code that looks like svg is actually jsx, that gets compiled to js and allows embedding values with the `{...}` notation.

So the mechanism to create new "skins", or actually more like alternative widgets, would be to subclass `DigitalComponent` and override its `render()` method. We could alter the widgets a little, so that for example unit handling code like https://github.com/SignalK/instrumentpanel/blob/b229e0beeab8a920e57cd59c142f3e5b36fe42d4/lib/ui/widgets/digitalcomponent.js#L46 would be part of the superclass and the subclass could only do the svg generation part.

Making this process dynamic, so that one could just drop new files on their own server and new widget variations would appear, is quite a bit of more work.

The easiest way to start with this would be that if you could just modify the DigitalComponent to your liking and issue a Pull Request with the changes. I and @cmotelet could then do the needed modifications so that it would be an addition of a new widget as a "sibgling" of current DigitalComponent and not a replacement.

Or if you have ideas for totally new widgets and you are handy with svg you could just post new issues with the svg attached and we could bring those into fruition included a live widgets in IP.

I think the viewBox was arbitrary or came from some grid layout settings, can't remember.

Displaying human readable labels is a job for the UI. The server can store this metadata for sharing on server or user level. The Signal K path is often needed for debugging, so it needs to be accessible.

tkurki, 2021-01-05:
Some ideas for widgets for different data:

https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/dashinst/dashinst.htmla

wmbutler, 2021-01-11:
As I've done more research, I almost feel like Grafana is the right approach. They've spent so much time on the dashboard configurations and time-series data integration that I almost think it makes more sense to develop custom widgets for that. I'm not trying to belittle your project in any way, but just working through the logistics of what to focus on.

I've got a pretty nice concept going on here and I think I'm going to see how heavy the lift is to create a widget in the grafana library to suit my purpose.

[image]

closed 2021-01-11 by wmbutler

tkurki, 2021-01-11:
No worries!

What db are you using in Grafana context? One problem I see is that it is usually not streaming: you need to keep refreshing the view, putting some strain on the backend and still not getting near real time data. I have a Proof of Concept of native Grafana datasource, that combines history retrieval over http with streaming SK data over websocket. Sadly unfinished.

Also one could possibly package the same gauges for consumptioin in Grafana, IP and the SK server admin UI. So please share what you come up with. And consider joining SK Slack community.