SignalK/instrumentpanel #175: Loading saved settings gives error

Fietsenwiel, 2021-01-26:
When I ltry to load the earlier saved settings I get an error:  

192.168.0.x:3000 meldt het volgende
an error occured in load of settings

It happens with all my devices (laptop, Smartphone, Tablet) and with different browsers (Edge, Chrome, InternetExplorer).

cmotelet, 2021-01-28:
The settings are saved on the server in a single file when you save for "save for global settings" located at : signalk home .signalk/applicationData/global/instrumentpanel/1.0.0.json or for "save for this user XXX" located at : signalk home .signalk/applicationData/users/XXX/instrumentpanel/1.0.0.json.
One of these files may be corrupted.
Can you send me the file to check it ?

Fietsenwiel, 2021-01-30:
Hi, thanks for your reply.

The 1.0.0.json look like this:

[1.0.0.json.txt](https://github.com/SignalK/instrumentpanel/files/5898282/1.0.0.json.txt)

My goal is to save the layout and restore it on a onther device. Now I know I also can save only the layout, that works well.

But another problem is tha I want to change the colors of the instruments (in the defaults.json file). That wos not an problem.
But when I change two devices it deosn't work anymore, see my default.json file:

[defaults.json.txt](https://github.com/SignalK/instrumentpanel/files/5898286/defaults.json.txt)

Whar it wrong with this defaults.json file?

Thanks in advance

cmotelet, 2021-01-30:
I restored your `1.0.0.json` and there are no errors in it.

I corrected your default.json. 
When you declare a path, it must be an object {} and not an array []  ( not `"energy": [`  ).
And you added an extra {} after energy.
[defaults.json.fixed.txt](https://github.com/SignalK/instrumentpanel/files/5898551/defaults.json.fixed.txt)
[image]

If you still can't reload your backup, you'll have to activate the debugger in chrome to get a little more information about the error.

Fietsenwiel, 2021-01-30:
Thank, you realy did help me. The instruments looks good now!
An for me is saving the layout good enough to load it on an other device.

Again, thanks!

cmotelet, 2021-01-30:
Is backup/restore working for you now or is there still an error?

cmotelet, 2021-01-30:
I misinterpreted your issue, I thought there was an error in the restore from the server. I just noticed that you were talking about restoring from a file. Indeed, yes there is an error.
I will correct it.
Thank you

cmotelet, 2021-01-31:
closed via #178 fixed in release in v0.17.2

closed 2021-01-31 by cmotelet

Fietsenwiel, 2021-01-31:
Hi,
Yes, i had also an issue with the restore of the local file (save all settings). And I still have that problem, j(ust tried).
For me this is not important anymore, saving "page layout" is for me fine now.

Thanks!

Fietsenwiel, 2021-01-31:
Oh, I think had had to update, I didn't. Try it later...

Fietsenwiel, 2021-01-31:
ok, tried ater updating and restarting but I still can't restore the "save all settings" locae file.

cmotelet, 2021-01-31:
The new release not yet publish on NPM only on Github.
we are currently working with Teppo to set up an automatic publishing process to NPM.
A little more patience ;-)

cmotelet, 2021-02-01:
The release 0.17.3 is now published on NPM with the fix.
You can update from the Signal K GUI