SignalK/instrumentpanel #176: Cross origin resource sharing problem with Firefox 84.0.2

norbert-walter, 2021-01-26:
When I start the instrument panel I see a empty page. The Web console means:

Uncaught DOMException: CSSStyleSheet.cssRules getter: Not allowed to access cross-origin stylesheet ui.js:52
    o index.mjs:29
    forEach self-hosted:206
    o index.mjs:28
    v index.mjs:55
    <anonym> main.js:35

In this case load the style sheets as inner link from other JS script. The server must allow the cross origin resource sharing. More details see hare:

https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS

norbert-walter, 2021-01-28:
Hello, the cross origin recource sharing problem ist not direct a problem from instrumentpanel. This is a web server problem. But it is important to know this.

Norbert

cmotelet, 2021-01-28:
I can't reproduce on Windows Firefox 84.0.1 & 85.0 last release.
Instrumentpanel is published/deployed with the source code (unminified js).
In the debugger tab, you can dig in the source code who raise the exeption and run step by step:
probably here:  webpack:///./node_modules/css-prefers-color-scheme/index.mjs.
[image]

I suspect a local addon in your browser calling an external URL or injecting an external CSS.

norbert-walter, 2021-01-29:
I discuss with Teppo in this issue about the same problem. I think the problem is in my network.
https://github.com/SignalK/instrumentpanel/pull/177#issuecomment-769251652

closed 2021-01-29 by tkurki