SignalK/instrumentpanel #183: Can't see any path

Bucaniere, 2021-02-28:
I'm able to see only white page with the words "Signal K Instrument Panel version 0.17.1". ican't see any paths. i've tried to reinstall OS and Signal K server but still nothing....
RPi4 with last openplotter

cmotelet, 2021-02-28:
Do you have errors in Chrome or Firefox browser (F12 key, then console content) when InstrumentPanel display the white page?
In Signal K server `Data Browser` page, select `self`, do you have paths and values ?
Can you provide the json content of  http://SignalK-server-IP:3000/signalk/v1/api/vessels/self/
replace SignalK-server-IP with localhost if your browser run on your RPi4 or with the ip of your Signal K server.
What is your Signal K server version ?

Bucaniere, 2021-03-01:
Thanks cmotelet.
- Seems to be an error in chromium (see pic).
- Typing what you suggest, I read “cannot GET /signalk/vi/api/vessels/self/“
- See pic of what I read in Signal K data browser 
- Signal K server version is 1.37.6

[image]

[image]

sbender9, 2021-03-01:
So you don’t have any data. Work on that.

cmotelet, 2021-03-01:
As Scott says, this is not an issue on InstrumentPanel.
If the Data Browser displays nothing, InstrumentPanel will have nothing to display.
Check your server entries in "Server/Data Connections".
You can also join us on Slack for more informations:
http://slack-invite.signalk.org/

cmotelet, 2021-03-21:
not InstrumentPanel issue

closed 2021-03-21 by cmotelet