SignalK/instrumentpanel #184: Panels showing after re config

Ianknx, 2021-03-17:
I have today set up a new installation of Signal K, I fully reformatted the SD card, re installed Open Plotter updated everything so a r start from scratch except on opening up instrument panel all the layout is populated with previous "
[image]
panels".

Is there no way to remove the layout and after completely formatting the sd card where are they coming from.

Regards

Ian

cmotelet, 2021-03-17:
these data come from your browser's local storage. To clean up :
[image]
Then `reset` at the bottom of the page.

Ianknx, 2021-03-17:
Wow as easy as that, I had used the reset for previous layouts earlier but
missed the delete all box.

Many thanks

Ian

closed 2021-03-17 by cmotelet