SignalK/instrumentpanel #200: Embed external HTML pages as panels

mairas, 2021-07-22:
As discussed on Slack: it’d be neat to be able to integrate external web apps to Instrumentpanel as iframes. That’d allow the users to embed e.g. grafana panels or dashboards, or other SK web apps to the Instrument Panel.

Considerations:
- Need UI for adding and deleting panels
- Need UI for giving custom names to panels

cmotelet, 2021-07-24:
The idea is good and @sbender9  had already suggested it to me but...
It would be a feature only in Instrumentpanel and not available for other webapps.
Moreover, it would add some code loaded in each client to manage the URL setup in a dedicated GUI and we go out of the scope where Instrumentpanel automatically discovers all the Signal K paths and displays them as well as possible.

I will propose you an alternative solution:
- a generic server plugin that send Signal K paths with URL resources
 `/vessel/self/external/<name>/{url: "http://myip:myport/myiframe"}`
 `external` or otherwise but a common path root to simplify processing on the client
- in Instrumentpanel a dedicated widget that displays the URL in a cell

This way:
- the resources are available to everyone
- no complex configuration to do on each client
- no dedicated GUI in the client code to enter parameters
- a very simple server plugin to do

There may be a problem of authentication, ideally the resource URL should be open access but read only

@tkurki & @sbender9 
what do you think about it ?

tkurki, 2021-07-29:
Imho iframe mechanism would be for wmbedding essentially random custom pages in IP. The ip layout the would be most probably be custom for embedding stuff and vice versa, the (for example grafana dashboard) page would be custom for working well with the ip layout.

So I see very little value in sharing the urls. Another plugin would make also installing this feature kinda complex for Joe Boater.

So I would implement this inside IP. Config forms per latout cell we have, but this would need a way to add new iframe cells.

Hmm. Now that I have written this far I started thinking about pushing Iframe urls as deltas and creating a gauge type for that and am having second thoughts. You could start with basedeltas. Imho we should have a basedelta editor in server admin - but even with that configuration would be pretty complex for Joe B...

cmotelet, 2021-07-30:
Thanks,
I'll start by trying to test a version with a baseDelta.json on the server to evaluate the rendering of an iFrame (Grafana) in an InstrumentPanel widget.
On the subject of a basic user, if he already has Signal K, InfuxDB, Grafana and some dashboards, I think he should be able to edit the baseDelta.json file, at least in the beginning.

cmotelet, 2021-08-11:
closed by #203

closed 2021-08-11 by cmotelet