SignalK/instrumentpanel #205: Loading error

mtl2500, 2021-10-09:
When I run the new version it will not login - the loading wheel just runs aroun
[image]

mtl2500, 2021-10-10:
Hi 

/signalk/v1/api/vessels/self/ = 

{"mmsi":"urn:mrn:imo:mmsi:219021302","name":"ANNA","class":"B","navigation":{"position":{"timestamp":"2021-10-28T10:50:40.000Z","latitude":55.490791,"longitude":9.728787,"source":"N2000-01"},"headingTrue":{"value":1.832502},"courseOverGroundTrue":{"value":0.000000},"speedOverGround":{"value":0.050000},"speedThroughWater":{"value":0.000000},"log":{"value":8912444},"logTrip":{"value":8912444},"magneticVariation":{"value":0.034900},"magneticVariationAgeOfService":{"value":65535}},"environment":{"wind":{"angleTrueGround":{"value":6.553506},"speedTrue":{"value":655.349976},"angleApparent":{"value":0.776601},"speedApparent":{"value":2.260000}},"outside":{"temperature":{"value":304.679993},"pressure":{"value":103400.000000}},"water":{"temperature":{"value":290.160004}},"depth":{"belowSurface":{"value":2.560000}}},"tanks":{"fuel":{"6":{"currentLevel":{"value":22}}},"freshWater":{"3":{"currentLevel":{"value":0}},"0":{"currentLevel":{"value":0}}},"blackWater":{"0":{"currentLevel":{"value":100}}}},"electrical":{"batteries":{"0":{"temperature":{"value":273.149994},"voltage":{"value":14.340000},"current":{"value":0.000000},"DCType":{"name":"battery"},"stateofCharge":{"value":0.000000},"stateofHealth":{"value":0},"timeRemaining":{"value":1}},"8":{"temperature":{"value":146.149994},"voltage":{"value":14.059999},"current":{"value":14.100000},"DCType":{"name":"battery"},"stateofCharge":{"value":0.000000},"stateofHealth":{"value":24},"timeRemaining":{"value":421}}}}}

mtl2500, 2021-10-10:
ui.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
site.webmanifest:1 Failed to load resource: the server responded with a status of 404 (Not Found)

Udviklerværktøjer kunne ikke indlæse kildekortet: Indholdet kunne ikke indlæses for http://192.168.0.103/hosted/instrumentpanel/css/bootstrap.min.css.map: HTTP-fejl: Statuskode 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

mtl2500, 2021-10-10:
﻿
ui.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)

site.webmanifest:1 Failed to load resource: the server responded with a status of 404 (Not Found)

Udviklerværktøjer kunne ikke indlæse kildekortet: Indholdet kunne ikke indlæses for http://192.168.0.103/hosted/instrumentpanel/css/bootstrap.min.css.map: HTTP-fejl: Statuskode 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
﻿
​

mtl2500, 2021-10-10:
The old version works fine -

cmotelet, 2021-10-16:
I checked with @tkurki  and it seems that your server is an iKommunicate. And so this version of InstrumentPanel is not usable with the Signal K server inside your gateway.

cmotelet, 2022-05-08:
InstrumentPanel running on iKommunicate is not supported with the latest versions.

closed 2022-05-08 by cmotelet