SignalK/instrumentpanel #206: Add conversion for inHg

tafischer, 2022-01-10:
Would it be possible to add inches of mercury to .pressure which is the standard measurement in the US?

cmotelet, 2022-01-10:
Yes, I will add this conversion in the next release.

pa1pdr, 2022-04-28:
Can I ask for a Joule to Wh conversion (for my battery capacities)? & whilst you're at it Coulomb -> Ah?

Greatly appreciated!

cmotelet, 2022-05-08:
@tafischer  & @pa1pdr done in release 0.22.1

tafischer, 2022-05-16:
@cmotelet - Hey hate to ask, but can you adjust the inHg to 2 decimal points?  I should have mentioned in the request, but we use two decimals.  ie 29.92

Thanks!

cmotelet, 2022-05-18:
@tafischer , No problem for the request, this is the place for it ;-)
It's true that I checked the conversion but I didn't ask myself if the precision of the display made sense.
The actual default rule for display numerical value is:
```
<10 => 2 decimal
>10 => one decimal
>50 => zero decimal
```
I will look at how to handle this particular case

tkurki, 2022-10-26:
Done i think

closed 2022-10-26 by tkurki