SignalK/instrumentpanel #232: Ceuld it be possible to remove path names in the display?

Mastiffen, 2022-10-26:
Since I use a phone as a display it takes up more space than I would like when the path names display. So if possible, I would like to have the option to just turn off the display of them.

tkurki, 2022-10-27:
Here is how it could look:
[image]

achieved by not rendering the widget label (path) and adjusting the main figure's font size and position accordingly
```
  render() {
    const showWidgetLabel = this.widgetLabel !== 'performance.velocityMadeGood'
    try {
      return (
        <svg height="100%" width="100%" viewBox="0 0 20 33" stroke="none">
          {showWidgetLabel && <text x="10" y="4" textAnchor="middle" fontSize="6" dominantBaseline="middle">
            {this.widgetLabel}
          </text>}
          <text textAnchor="end" x="62" y="30" fontSize={this.unitFontSize} dominantBaseline="baseline">{this.finalUnit}</text>
          <text textAnchor="middle" x="10" y={showWidgetLabel ? 22: 20} fontSize={showWidgetLabel ? 29 : 35} dominantBaseline="middle">
            {this.state.value.toString()}
          </text>
        </svg>
      )
    } catch (ex) { console.log(ex) }
    return (<div>safety mode</div>)
  }
```
This was just a quick poc. For real this could be one more checkbox in the digital widget's settings. 

Or IP could in addition hide the the label if displayName is set to empty string - this would have global effect: if you don't want to see the label you can set the displayName to the empty string.

Mastiffen, 2022-10-27:
That's exactly what I meant, thanks! So the numbers take up more space, and I can either have them bigger or the same size, with room for more data! And I must say I'm all for checkboxes, it would make it very easy for me.

cmotelet, 2022-10-27:
Yes, that's how I imagined the solution. 
A "hide label" checkbox seems to me the most neutral solution.
The displayName to an empty string will impact all clients and that's a bit extreme.
I'll put this in a state to avoid multiple switches `showWidgetLabel ? X : Y` on each call to render().
I'm also thinking of making the labels reappear when the menu bar becomes visible again.

tkurki, 2022-10-28:
Showing them with the menubar logic is a good idea!

closed 2022-12-04 by cmotelet