SignalK/instrumentpanel #251: save settings bug

tonipezzoni, 2023-09-14:
The m/s unit is always saved, even if a different unit was choosen. The problem concerns both single user and global.

cmotelet, 2023-09-15:
Can you give me a little more detail and the steps to reproduce ?
In the saved file, when you restore, in the preferred units, on a particular widget?
In file configuration, in unit, there is always `m/s` because it's the default unit on path source.
But in `convertTo` th is your display unit:
`
{
"type": "universal",
"options": {
  "sourceId": "vcan0.43",
  "key": "vcan0.43.navigation.speedOverGround",
  "path": "navigation.speedOverGround",
  "active": false,
  "selectedWidget": 0,
  "label": "navigation.speedOverGround",
  "unit": "m/s",
  "convertTo": "kn"
}
`

cmotelet, 2024-09-18:
Closed, no response for over a year