SignalK/instrumentpanel #253: Could it be made possible to adjust width in smaller increments?

Mastiffen, 2023-12-23:
I have a phone at the front of the boat, where I set and pull my crawfish and lobster pots. That way I can see depth and some other good to have info without going to the plotter in the steering house. The problem I have is that I would like some info to be two in one line in portrait mode on the phone. But I can't split them down the middel, on the IP67 Sony phone I use there I can only have 1/3 of the width or 2/3 of the width, and that means that one of the panels are cut off. Is it possible to have smaller increments in the adjustment for that? Or is there a setting that isn't documented in the manual about resizing?

cmotelet, 2023-12-23:
The number of possible columns and their pixel sizes are automatically calculated according to screen size. There's no way to configure it manually.
Can you send me a screenshot and the pixel size of your screen ?

Mastiffen, 2023-12-24:
Thank you for the quick answer!  It's a Sony Xperia 10 II, which has a 21:9 screen, maybe that's the problem? 1080x2520 pixels. As you can see the water temp is almost unreadable, and the front voltage is taking 2/3, while the rerr voltage  has to settle for 1/3.
[image]

Mastiffen, 2024-01-05:
This may be a moot point for me soon. I am just waiting for the new Oneplus 12, then I will move my current main phone, Huawei P30 Pro, to display use in the boat. I assume the different display size will not have the same problem as the Sony.

Mastiffen, 2024-02-17:
I found out that I have the same problem on the Huawei, and I also found the reason for the problem: The width is not calculated correctly when the label is not shown. Totally repeatable. When the label is there, the full number is within the window, when the label is not there, the font is too big and is cut off vertically.

cmotelet, 2024-02-21:
Indeed, I've added a fixed zoom factor when the label is hidden to make the most of the space freed up. In your case, perhaps you shouldn't hide the label. I'll try to add the possibility of adjusting the zoom factor in the widget configuration.

Mastiffen, 2024-02-21:
Thanks, and good, I'm glad I'm not nuts and hallucinating this! How about another thing? I hide the labels because they don't tell my wife anything when she's at the helm. If i could give them actually descriptive names instead of the fixed data path, that would be even better than hiding.

cmotelet, 2024-02-22:
you can give names to paths, but this is not done in intrumentpannel but on the server itself. it's explained here:
https://github.com/SignalK/signalk-server/wiki/FAQ:-Frequently-Asked-Questions#how-to-add-missing-units-and-static-data-eg-displayname

Mastiffen, 2024-02-22:
Wow, I was looking in the plug-in settings! I tried with a couple of values now, and that worked perfectly. Thanks!