SignalK/instrumentpanel #254: Instrument panel forgets settings

wcbonner, 2024-02-11:
I'll go through configuration of the panel the way I want it, leaving it open as a tab in my iPhone, and a random number of days later the tab will be reverted to a default setting with different widgets displayed and the default speed shown as m/s instead of the kts that I prefer. [image]

cmotelet, 2024-02-12:
I have no explanation for this behavior.
If you setup your configuration, wait a few minutes and close the page on your iPhone. When you reopen it, you see the same behavior (loss of settings and layout) ?

tkurki, 2024-02-12:
See 7 day limit on script writable storage https://webkit.org/blog/10218/full-third-party-cookie-blocking-and-more/

wcbonner, 2024-02-13:
> See 7 day limit on script writable storage https://webkit.org/blog/10218/full-third-party-cookie-blocking-and-more/

The 7 day limit could easily be my problem. My thought is that if it's a cookie expiration issue, the web page should update its cookie expiration every time the tab is active. 

I mainly opened the issue because losing the settings regularly discourages me from using the page, and I wanted to find if other people have experienced this same problem, or what might be different about my usage that causes it.

tkurki, 2024-02-13:
@cmotelet we could change IP so that it stores and automatically retrieves the settings and layouts from the server, keyed on the screen size so that different size devices would have their own layouts.

cmotelet, 2024-02-13:
@wcbonner ,
In quick win, you can add InstrumentPanel to the home screen so as not to be affected by the purge after 7 days. This is explained in the online help section `4.2.1 On iPad or iPhone` . the procedure may be slightly different depending on the IOS version.

wcbonner, 2024-02-16:
> @wcbonner ,
> 
> In quick win, you can add InstrumentPanel to the home screen so as not to be affected by the purge after 7 days. This is explained in the online help section `4.2.1 On iPad or iPhone` . the procedure may be slightly different depending on the IOS version.

I had not thought of mentioning that I'm using Google Chrome. Also I don't need more icons on the screen of my iPhone. Still, I may try it just to see if it doesn't forget the settings after a short period. 

https://demo.signalk.org/@signalk/instrumentpanel/#4_2_1

> 4.2.1 On iPad or iPhone Back
>
> Launch Safari app. This does not work from the Chrome app.

wcbonner, 2024-02-16:
It's interesting to note that following the link to the demo site brings up the default layout that I am trying to avoid reverting to. 

[image]

wcbonner, 2024-02-19:
At least I know that it's the seven day limit now. This morning when I looked at the instruments I got this page: [image]

Then after I dismissed it, it went back to this page with instrumentation I didn't want. [image]

cmotelet, 2024-02-20:
have you also tried the same thing by adding it to the home screen ?

wcbonner, 2024-02-20:
> have you also tried the same thing by adding it to the home screen ?

I've not tried adding it to the home screen. 

I think I read that feature is Safari specific, as well as possibly going away with the next version of iOS. 

I just went through the steps, the new page needed to be configured fresh, so I did that and I'll see what it's reading a week from now. 

[image] [image] [image] [image] [image] [image]

cmotelet, 2024-02-20:
This is the normal behavior, the version on the home screen is considered by the device as a new site. 
This won't fix the 7-day problem, but to avoid redoing your layout every time, you can save it on the server: 2.3.5. Save settings screen

wcbonner, 2024-03-06:
The launched page from the stored icon seems to remember its settings. It's now 23 days since I started this issue, so probably two weeks since I tried the solution, including the iOS 17.4 update.

closed 2024-03-06 by wcbonner