SignalK/instrumentpanel #266: Layout configuration lost, and no way to load it back

Krecik7, 2024-08-18:
Hello, i have similar problem on two installations of SignalK/InstrumentPanel, last i think two weeks when it stopped to work - 

First - both installations forget saved configuration and reset to default. I use them everyday so i don't think to have cookie problem, just one day in morning when i open IP i see default configuration. 

BUT

Both are unable to load saved layout configuration. No matter if i try to do this on Windows (Firefox/Chrome) or on Android, both show only ENABLE/RESTORE/RESET, but with 'default' layout on page. 

[image]

I think - this can be something on SignalK side, working good till now. But i don't have idea, where to start looking for solution. 

All old saved layouts are useless right now, even if i do new layout, save it (as admin -> global) and then try to restore - then nothing is changing.

cmotelet, 2024-08-18:
There's a bug in the restore function, I'll fix it as soon as possible.

cmotelet, 2024-09-18:
Fixed in release `0.26.0`

closed 2024-09-18 by cmotelet