SignalK/instrumentpanel #280: Cannot resize or arrange gauges in Firefox Browser

boatmon0, 2024-10-22:
I see an item with this title closed in 2016.  I currently have this problem on the firefox browser.  It works if I use Chrome.  I'm thinking that it might be a setting in Firefox?  Some cookie kind of issue.  When I edit a panel, and rearrange the gauges, they stay in the chosen place in the editor.  When I close the editor, the return to the old position.  If I open the editor, even after a reboot,  the gauges are where I placed them.  

I'm sure my Firefox is up to date.  I'm sure instrumentpanel is up to date.  I run keepass browser extension, but that's about it.  I have Firefox set to "standard" security level.

Standard
Balanced for protection and performance. Pages will load normally.
Firefox blocks the following:
Social media trackers
Cross-site cookies in all windows
Tracking content in Private Windows
Cryptominers
Fingerprinters
Includes Total Cookie Protection, our most powerful privacy feature ever
Total Cookie Protection contains cookies to the site you’re on, so trackers can’t use them to follow you between sites.

One more additional thing, on firefox editing my wind instrument looks like this:
[image]

And on Chrome it looks like this:
[image]

Firefox is not populating values from the wind instrument.  Looking at the data, view on signalk the data is being displayed properly.  Clearly something is up with Firefox.  Is is a cookie issue?

Thanks for any suggestions!

cmotelet, 2024-10-22:
Sorry but I can't reproduce with my configuration:
Signal K: `2.11.0`
InstrumentPanel: `0.26.0`
Firefox: `131.0.3 (64 bits)`
OS: `Windows 11`

Can you activate the debugger (F12 key) and check whether error or warning messages appear in the console tab ?

InstrumentPanel does not use `Cookies` but only `Storage \ Local Storage`.
[image]
[image]

boatmon0, 2024-10-22:
Thanks cmotelet!
I've fought this for some time.  Finally late last night, I decided to delete all of my cookies.  It was 1.2gb of cookies.  Now Firefox does allow me to rearrange gauges.  I do however still have a problem with the wind gauge?

I sure looks like the gauge rearrange issue is a Firefox issue.

Thanks!